SWC-Advanced-Microscopy/multiphoton-qc #86: Get the gains to acquire for the standard source from the lens paper data

raacampbell, 2025-02-15:
Look in the lens paper data from the current directory and extract the PMT gains from each file. Then acquire at the same gains with the standard source. This will avoid us having to specify any gains at all at the CLI and it will make the hard-coded gain series file in the `private` directory for the `record` functions less critical.

sumiya-kuroda, 2025-02-24:
I think `mpqc.record.lens_paper` right now saves file with only one PMT gain while the following analysis pipelines such as `mpqc.analyse.get_quantalsize_from_file` assumes there are files with different gains.

raacampbell, 2025-02-26:
It does not assume multiple gains, I don't think, but it does look for the gain in the file name. I have not sorted this out yet. I think there is no strong need to do the lens paper at a range of gains, but it should be done at the same gain for all detectors.