SWC-Advanced-Microscopy/multiphoton-qc #93: Lagging ScanImage after using mpqc.tools.meanFrame

sumiya-kuroda, 2025-02-26:
After using and even closing `mpqc.tools.meanFrame`, ScanImage remains very laggy. Restarting the ScanImage fixes the problem for now.

raacampbell, 2025-02-26:
This is really interesting because I have seen this behavior in other scenarios with ScanImage but could not reproduce it. Is it reproducible for you? The only thing I can think is that the listeners are not deleted. See if this commit on `dev` fixes things. cd4f23b9982b5b9218a55bfa3de24a5da2e0469f

raacampbell, 2025-02-27:
> No more lag after the fix! Thanks so much!

Great. I think the same issue might be happening on the serial section
software, but the codebase is huge and it happens only rarely so I have no
idea where to look. At least now I have something to look for!

sumiya-kuroda, 2025-03-25:
Today, with the latest version of the repository, I encountered another issue. While I use `mpqc.tools.meanFrame`, ScanImage itself is not lagging anymore but the Channel 1/2 images are. For example, when I move my XY stange, the Channel 1 and 2 images start to move after >1 second of delay. This behavior stops and the images become responsive once I stop the function.

raacampbell, 2025-03-25:
Are you sure this is new? Nothing in the code has changed so maybe it always did this. I would suggest looking at the `updatePlot` method, as this is the one that runs on each frame, and see with commenting out lines what is taking the time. You could also contact ScanImage and see if they have a suggestion for doing this better. e.g. maybe a timer that does the update in a different thread.  If the issue persists and you want to comment more, start a new Issue as I think what you are seeing here is distinct from the original problem we fixed in this Issue.